ronibandini/tallerPocketGone #2: No funciona led

zapateraz, 2025-01-30:
Buenas, me pasa que tengo todo en instalado y los cables bien conectados pero no me prende el led, alguna idea de que podrá ser?
Gracias.

https://github.com/user-attachments/assets/72d33c22-c20a-42f7-97d8-d56979c399f1

ronibandini, 2025-01-30:
Hay un error en la imagen del video sobre el circuito de botón y LED. La pata más larga del LED debe ir conectada a GPIO y la más corta al GND (puenteado con el interruptor). 

[image]
[image]

Perdón por la confusión

closed 2025-01-30 by ronibandini

zapateraz, 2025-02-01:
https://github.com/user-attachments/assets/ddf409c8-eba6-491b-af1b-9a1b1dce70ec

Así debería de funcionar?

ronibandini, 2025-02-01:
Si estás usando un botón con retención, al prender y apagar debería encender la unidad (y el LED) 
Para que deje de operar nuevamente habría que prender y apagar. 

Con el botón recomendado se toca una vez para prender y una para apagar.

Lo mejor es que lo pruebes con un equipo bluetooth cerca

zapateraz, 2025-02-01:
No, es un boton sin retención como si fuese un pulsador. Lo compré tal como dice el tutorial, el mismo modelo inclusive.