ronibandini/tallerPocketGone #3: No funciona botón

zapateraz, 2025-01-30:
Ahora prende el led pero no funciona el botón, que podrá ser?
Gracias

https://github.com/user-attachments/assets/2e241399-c8e7-471b-9a6c-282cb1fd9bf2

ronibandini, 2025-01-30:
Verificá que el botón tenga un extremo a GPIO33 y el otro extremo conectado a GND. Hay algunos pulsadores económicos que no hacen bien contacto y hay que moverlos al presionar. También podés abrir el monitor serial del Arduino IDE en 115200 baud rate y verificar los mensajes que aparecen

ronibandini, 2025-01-30:
Más información sobre Monitor Serial: conectan la placa con el cable microUSB, abren Arduino IDE, presionan Herramientas, Monitor Serial o en Windows CTRL Shift M

zapateraz, 2025-01-31:
https://github.com/user-attachments/assets/efc79e9c-4255-4005-94a6-a7d3881f6784

Asi puede llegar a funcionar igual?

ronibandini, 2025-01-31:
Podría funcionar, pero creo que deberías prender y apagar para simular el efecto del botón sin retención (en lugar de dejarlo en la posición de prendido)

zapateraz, 2025-02-01:
https://github.com/user-attachments/assets/17be4220-ee90-4ea7-b717-8bc64bec76a8

Asi deberia quedar?

ronibandini, 2025-02-01:
¿Tenés algo conectado a GPIO25?

zapateraz, 2025-02-01:
No, esta en puente gnd con gpio 0 y gpio 32 el boton.

ronibandini, 2025-02-01:
El boton va a 33 y GND. El led va a 32 y GND.

zapateraz, 2025-02-01:
https://github.com/user-attachments/assets/8db97e4b-ddc4-4871-83a2-c84ed1a10eb4

Me pasa que el boton no funciona y de repente se prende y apaga solo

https://github.com/user-attachments/assets/7fa396eb-e2ed-4038-8cc3-688989c04a3b

ronibandini, 2025-02-02:
El dispositivo tiene un timer. 45 segundos opera. 15 segundos se deshabilita. Si el botón no está funcionando tenés que verificar estar conectando los GPIO correctos. En distintas placas pueden estar en otros lugares. La otra opción es el modo jumper, ya explicada, que no require del botón y se enciende automáticamente.

closed 2025-02-02 by ronibandini

jaynqn, 2025-03-08:
Hola, estoy teniendo el mismo problema, no enciende el LED. 
El boton que compre viene con cables soldados ya así que conecté desde sus puntas.
Las patas de los LED que compré son casi del mismo largo, probé intercambiar la conexión de una a la otra pero tampoco enciende el LED.
Probé con otro botón del mismo tipo también.
Abrí el Arduino IDE pero no se que buscar ahi. 

[image]

[image]

jaynqn, 2025-03-08:
Donde explicaste el modo jumper?

ronibandini, 2025-03-08:
¿En qué etapa del software estás? ¿Llegaste a subir el software .bin? Acá podés encontrar una guía de polaridad prescindiendo del largo de las patas del LED https://www.switchelectronics.co.uk/blogs/news/led-polarity-basic-guide

jaynqn, 2025-03-09:
Hola Roni, funciona! Era el soft. Hice el update otra vez y empezó a andar.
Acá va un video.

https://github.com/user-attachments/assets/de3dc27a-e475-4b11-bd16-5e0ebc4746d1

Estaba poniendo la música desde el teléfono que filma así que bastante cerca del parlante, pero tuve mejores resultados cuando lo alejé un poco.

Me quedan algunas preguntas nomás:
1. Que tecnología de impresión 3D uso? Me ofrecen FDM, MJF, SLA, SLS, etc.
2. Voy a necesitar un poco más de alcance, sin entender sobre el tema, las opciones que veo son:
     a. Comprar una antena de router como sugeriste en el foro (por ejemplo https://shorturl.at/i1gdf). Aunque no puedo comprobar que sea más potente que la incluida con la placa.
     b. Comprar una antena SMA Dipole como dice en el manual (por ejemplo https://shorturl.at/H6CFg)
     c. Comprar algo así https://shorturl.at/9iRzO aunque no se si me podré meter en problemas si resulta muy potente
3. Para hacer un ataque quirúrgico, conociendo el mac address del dispositivo, sería necesario cambiar el código, no? Vi que el RBG funciona así.

Muchas gracias,

Juan

ronibandini, 2025-03-09:
Excelente, Juan.

1. FDM con filamento PLA va bien
2. Recomiendo probar primero con antenas - un chico en Uruguay extendió al doble el alcance apenas con una antena - y sino meter en el medio un amplificador de 2.4, pero siempre hay que contemplar que depende de varios factores: ubicación de emisor, receptor y Pocket Gone, parlante utilizado y obstáculos
3. Exacto