ronibandini/tallerPocketGone #5: Archivos 3D

Elsebito007, 2025-01-30:
Cuando puedas, me pasas los archivos 3D asi averiguo cuento cuestan?

ronibandini, 2025-01-30:
Hola. En Argentina ambas piezas en PLA tienen un costo de $3000 aprox Los archivos gcode se envían junto al BIN una vez reportado el MAC.

closed 2025-01-31 by ronibandini

Elsebito007, 2025-02-11:
Hola! Tengo una Mac. Llegue hasta que conecto la placa y se prende la luz roja que tiene. 
Ahora me pasa esto: Estoy en el Arduino, en una parte del video me muestra que en el Arduino deberia ver lo sgte: 

[image]

Pero lo que veo es esto:

[image]
 Probe seleccionando cada uno, pero cuando tipeo "DEV" no me aparece lo que se ve en esta imagen:

[image]

Estoy haciendo algo mal? Como sigo?

ronibandini, 2025-02-11:
En Mac es probable que existan algunas diferencias. Probá la opción de usbserial y si no funciona fijate algún tutorial de Arduino IDE para Mac https://www.youtube.com/watch?v=6eMyKhAx--g

Elsebito007, 2025-02-11:
> En Mac es probable que existan algunas diferencias. Probá la opción de usbserial y si no funciona fijate algún tutorial de Arduino IDE para Mac https://www.youtube.com/watch?v=6eMyKhAx--g

No veo que en el video haya alguna solucion a lo que me pasa. 
Puedo cerrar el Arduino y desconectar la placa sin que pase nada? Puedo volver a intentar esto en una PC desde cero? Todo esto es chino para mi.

ronibandini, 2025-02-11:
Si. No hay problema.