ronibandini/tallerPocketGone #8: Antena

zapateraz, 2025-02-03:
Consulta, si se le pone una antena mas grande ¿Tiene mejor alcance o hay que configurarle algo más a alguna de las placas o al software?

Gracias

ronibandini, 2025-02-03:
Hasta cierto punto una antena superior te va a extender el alcance. Luego ya tendrías que evaluar el uso de un amplificador entre el Pocket y la antena.

Grinch1983, 2025-02-03:
Hola Roni. Al subir el sketch me tiró ese error. no soy muy entendido del tema de programar por eso te pregunto si está bien dado que me pude conectar al wifi de la placa y sacar su MAC address. Desde ya muchas gracias

D:\Descargas D\getMacOta\getMacOta.ino: In function 'void setup()':
D:\Descargas D\getMacOta\getMacOta.ino:18:8: error: 'class WiFiClass' has no member named 'STA'
   WiFi.STA.begin();
        ^~~
D:\Descargas D\getMacOta\getMacOta.ino:21:8: error: 'class WiFiClass' has no member named 'STA'
   WiFi.STA.end();
        ^~~

exit status 1

Compilation error: 'class WiFiClass' has no member named 'STA'

ronibandini, 2025-02-03:
Hola. Si te pudiste conectar y sacar el MAC, eso significa que el sketch subió correctamente a la placa. Podés pedir el software y seguir adelante con los pasos-

closed 2025-02-05 by ronibandini