ronibandini/tallerPocketGone #9: Doit Dev Kit V1 no aparece en la lista de boards

xaavito, 2025-02-03:
Realize todos los pasos pero no encuentro ese item en la lista, de hecho en el video se ven algunas que a mi no me aparecen
Estoy usando una mac

[image]

ronibandini, 2025-02-03:
Si cargaste en preferencias, placas adicionales el URL https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json, probá cerrar y volver a abrir el IDE

En caso que siga sin aparecer, probá seleccionar ESP32 Dev Board o alguna que diga ESP32S

xaavito, 2025-02-03:
Despues de seleccion es32 dev board bajo drivers y encontro del dev kit que correspondia con el video, erstoy intentanto subir el sketch
pero me tira

```
A fatal error occurred: MD5 of file does not match data in flash!
Failed uploading: uploading error: exit status 2
```

closed 2025-02-04 by xaavito

grafixmo, 2025-02-09:
Me pasó lo mismo. Se solucionó al reiniciar el Mac. 

Un segundo problema fue que tuve que bajarme un drive PARA la ESP32 de Mac OS  para finalizar la instalación del software en la placa:
[](https://www.silabs.com/developer-tools/usb-to-uart-bridge-vcp-drivers)

Reiniciado de nuevo el Mac, todo fue  bien.

mrocach, 2025-03-23:
Hola. Lo dejo en comentarios para quien le sirva. 
Inicialmente seleccioné el de "Arduino ESP32". Cargue el sw y me dio error en las librerías de WiFi.
Reinicié y nada. Como no me aparecía, seleccioné uno que estaba "deshabilitado" que decía "ESP32 Dev Board". Le di OK y me preguntó si quería instalar la librería. le dije que si (demora un poco). Una vez instalado, vuelves a la pantalla inicial del arduino, seleccionas nuevamente la placa, buscas el "DOIT ESP32 DEVKIT V1" y ya te aparece en el listado.

[image]

ronibandini, 2025-03-24:
Gracias por publicar los detalles