ronibandini/tallerPocketGone #10: A fatal error occurred: MD5 of file does not match data in flash

xaavito, 2025-02-04:
Despues de poder seleccionar bien el board empiezo la instalacion y me tira esto

```
Sketch uses 937688 bytes (71%) of program storage space. Maximum is 1310720 bytes.
Global variables use 46204 bytes (14%) of dynamic memory, leaving 281476 bytes for local variables. Maximum is 327680 bytes.
esptool.py v4.8.1
Serial port /dev/cu.usbserial-0001
Connecting....
Chip is ESP32-D0WD-V3 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 48:e7:29:b6:8a:88
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00006fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x000f4fff...
Compressed 23440 bytes to 15065...
Writing at 0x00001000... (100 %)
Wrote 23440 bytes (15065 compressed) at 0x00001000 in 1.9 seconds (effective 100.7 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 146...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (146 compressed) at 0x00008000 in 0.1 seconds (effective 246.3 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.2 seconds (effective 394.7 kbit/s)...
Hash of data verified.
Compressed 937792 bytes to 596677...
Writing at 0x00010000... (2 %)
Writing at 0x0001bfd2... (5 %)
Writing at 0x00026956... (8 %)
Writing at 0x00031801... (10 %)
Writing at 0x0003dc5f... (13 %)
Writing at 0x000438e3... (16 %)
Writing at 0x000497b5... (18 %)
Writing at 0x0004efaf... (21 %)
Writing at 0x00054231... (24 %)
Writing at 0x00059ae4... (27 %)
Writing at 0x0005eda9... (29 %)
Writing at 0x000641d1... (32 %)
Writing at 0x00069a89... (35 %)
Writing at 0x0006f017... (37 %)
Writing at 0x00074936... (40 %)
Writing at 0x0007a055... (43 %)
Writing at 0x0007f2d9... (45 %)
Writing at 0x000844e9... (48 %)
Writing at 0x00089f1d... (51 %)
Writing at 0x0008f87d... (54 %)
Writing at 0x00095331... (56 %)
Writing at 0x0009acd0... (59 %)
Writing at 0x000a01e3... (62 %)
Writing at 0x000a550f... (64 %)
Writing at 0x000aaa06... (67 %)
Writing at 0x000afd05... (70 %)
Writing at 0x000b5400... (72 %)
Writing at 0x000baa57... (75 %)
Writing at 0x000c03c3... (78 %)
Writing at 0x000c612d... (81 %)
Writing at 0x000cba7f... (83 %)
Writing at 0x000d6bd3... (86 %)
Writing at 0x000dc80f... (89 %)
Writing at 0x000e1906... (91 %)
Writing at 0x000e715c... (94 %)
Writing at 0x000ec924... (97 %)
Writing at 0x000f26af... (100 %)
Wrote 937792 bytes (596677 compressed) at 0x00010000 in 52.6 seconds (effective 142.5 kbit/s)...
File  md5: 294712a03cdbe609f440eeb0d2f177f7
Flash md5: a300580d4c629f750495a825ce227ec8
MD5 of 0xFF is bcdd8d528d458901fcc29535333d81d8

A fatal error occurred: MD5 of file does not match data in flash!
Failed uploading: uploading error: exit status 2
```

probe un par de sugerencias de la red como bajar la velocidad o usar ciertos comandos como estos, pero sin exito, siempre el mismo error

```
python3 -m esptool --chip esp32 --port /dev/cu.usbserial-0001 --baud 115200 --after hard_reset erase_flash

python3 -m esptool --chip esp32 --port /dev/cu.usbserial-0001 --baud 115200 write_flash_status --non-volatile 0
```

ronibandini, 2025-02-04:
Algunos indican que el problema se puede deber al cable USB utilizado o ESP32 defectuosos. Este fix le funcionó a varios

https://community.m5stack.com/topic/838/a-fatal-error-occurred-md5-of-file-does-not-match-data-in-flash/5

xaavito, 2025-02-04:
Gracias ya lo pruebo!
te comento que por las dudas cambie de maquina a un ubuntu, y probe subirle un hello world y anduvo, el bardo viene con algo de la config de este proyecto estimo
descarto el cable y el esp32 que este fallado

xaavito, 2025-02-04:
Probe ejemplos que vienen por ddefecto, algunos los compila bien y otros no
tengo que ir por la idea que hay algo mal en el esp32?

ronibandini, 2025-02-04:
En principio si podés subir el getMacOta.ino, avanzá con la instalación

xaavito, 2025-02-04:
Falla con el error del md5 y jamas encuentro el wifi.. estimo jamas termino de instalar correctamente

ronibandini, 2025-02-04:
En ese caso podés probar desde otra computadora y si no funciona cambiar el ESP. En caso que tengas algún otro ESP32 seguramente se pueda usar comprobando usar los pines correctos

xaavito, 2025-02-04:
Probe con win y ubuntu y el mismo caso
lo raro es que algunos ejemplos si funcionan y otros no, con eso descarte el cable y los puertos de la pc
ya me mande a comprar otra esp32 y veremos

xaavito, 2025-02-04:
No hubo caso, compre otro esp32 y este anduvo de 10 de una, cierro

closed 2025-02-04 by xaavito

ronibandini, 2025-02-04:
Excelente. Exitos con el cableado y el circuito.