ronibandini/tallerPocketGone #11: No desconecta el parlante jbl

Malanoche593, 2025-02-07:
Hola, por favor me pueden ayudar hice todos los pasos y al principio trato de silenciar el parlante pero después ya no lo hace. El LED prende y se apagada según la funcim del boton.

[image]

Malanoche593, 2025-02-07:
Probé con 2 módulos de NRF pero no me hace efecto y las conexiones del módulo con el esp32 está como indica en las pasos de ensamblaje.

[image]

[image]

ronibandini, 2025-02-07:
¿Probaste con otros parlantes bluetooth?

ronibandini, 2025-02-07:
En caso que hayas probado con otros parlantes BT cerca y no los silencia, podés conectar el cable microUSB del ESP32 a la computadora, abrir Arduino IDE, Serial Monitor en 115200 y ver los mensajes de diagnóstico para confirmar que la placa de radiofrecuencia se inicialice correctamente

Malanoche593, 2025-02-08:
Roni al principio si intentaba silenciar el parlante porque la musica comenzó a sonar con intermitencia  pero ahora no si quiera lo intenta hacer volví a conectar todo de nuevo inclusive directamente pero sigue sin hacer su función.  El monitor serial a continuación. 

[image]

[image]

Malanoche593, 2025-02-08:
[image]

Malanoche593, 2025-02-10:
No tengo más parlantes solo un jbl flip 4 no se porque al principio si lo intentaba silenciar pero ahora la música suena sin problema

ronibandini, 2025-02-10:
Para diagnosticar el problema conviene que pruebes con al menos 3 parlantes Bluetooth. Algunos parlantes recientes podrían no ser vulnerables.

Malanoche593, 2025-02-10:
Estaba intentando también con unos airpods conectados con la PC pero tampoco corta la música.

Malanoche593, 2025-02-11:
Hola, ya pude resolver la falla que no silenciaba el parlante, tuve que ir a la tienda donde lo compre para que me cambien la antena del módulo dado que se veía de otra manera a la antenas que normalmente venden. Una vez cambiado al fin puedo silenciar el parlante.

Malanoche593, 2025-02-11:
Adicional estoy utilizando el Esp32 wroom 32U.

ronibandini, 2025-02-11:
Gracias por reportarlo. Nunca se me hubiera ocurrido que vendieran la placa de RF con la antena equivocada.

closed 2025-02-11 by ronibandini