ronibandini/tallerPocketGone #12: No conecta ip para subir el archivo bin. Necesito ayuda

Nvaccaro1970, 2025-02-07:


Nvaccaro1970, 2025-02-07:
Realizo la coneccion de la red que en mi caso es ESP_FCDEF5, en el navegador tipeo http://10.10.10.1/update y no me abre nada para cargar el bin

ronibandini, 2025-02-07:
Hola. Parece que te estás conectando a algún otro lado. La conexión WiFi debería llamarse PocketGone00:00:00:00:00 (En lugar de 00... el Mac Address de tu placa)

Nvaccaro1970, 2025-02-07:
No estaba instalado, porque no reconocia el usb, por eso no funcionaba, instale el driver que me aparecia cuando conectaba la placa  cp2101 usb to uart bridge controller y lo pude cargar correctamente. Lo que necesito ahora es nuevamete el bin porque la mac no es la misma. decime el procedimiento para obtenerlo

ronibandini, 2025-02-08:
El MAC address de la placa no cambia por haber instalado drivers, de modo que deberías poder usar el mismo .bin que ya fue compilado y enviado. En caso que hayas cambiado de placa o hayas indicado un MAC address equivocado es necesario recompilar el software y eso tiene costo.

Nvaccaro1970, 2025-02-08:
Si la Mac no es la correcta que habia indicado, como te habia puesto arriba me aparecia una red ESP_FCDEF5 y no con la MAC, lo busque en propiedades pero sin dudas no era esa, no prende ni el led, ni le llega voltaje. Decime como procedo para un nuevo bin?

ronibandini, 2025-02-10:
Si fue un error tuyo al reportar la MAC podés encargar nuevamente el software en https://www.patreon.com/RoniBandini/shop/pocket-gone-cambio-de-mac-address-1050139

En caso contrario mandame un email.