ronibandini/tallerPocketGone #14: Necesito ayuda

turcoaquim, 2025-02-09:
Hola Roni logre entrar aqui para la consulta, hago todo hasta conectar la placa por usb y bajarle el soft pero cuando voy a las conecciones de wifi no me aparece la opcion pocketgone y el numero siguiente. que podria ser? no puedo avanzar gracias

ronibandini, 2025-02-10:
Algunos usuarios reportaron necesitar un paso previo a subir el software getMacOTA.ino que consiste en instalar los drivers cp2101 usb to uart bridge controller

turcoaquim, 2025-02-10:
Hola, chino basico para mi Roni discukpa, bajar que drivers de donde e instalarlos como? Gracias

ronibandini, 2025-02-10:
Acá hay un tutorial paso a paso https://www.youtube.com/watch?v=Hcj1CokUa68

leandro7gomez, 2025-02-10:
Buenas tardes Roni! ya compré el curso! que tipo de boton debo pedir para no estarlo presionando todo el tiempo? Y cual antena para tener mayor alcance?  la placa Nrf24l01 tiene el mismo pin para enroscar la antena que los routers? gracias.

ronibandini, 2025-02-10:
El diseño original no requiere estar presionando el botón permanentemente. Se toca una vez para prender. Luego podés apagarlo y prender el segundo modo. Podés probar antenas externas de router, pero fijate que vengan con el pin SMA macho como la antena original de la Nrf24l01

leandro7gomez, 2025-02-10:
> El diseño original no requiere estar presionando el botón permanentemente. Se toca una vez para prender. Luego podés apagarlo y prender el segundo modo. Podés probar antenas externas de router, pero fijate que vengan con el pin SMA macho como la antena original de la Nrf24l01

Gracias! Cuando tenga la plaquita te escribo en el formulario... Saludos desde Mendoza.

closed 2025-02-11 by ronibandini