ronibandini/tallerPocketGone #15: Placa ESP32DEVKit V1

fercava, 2025-02-10:
Hola Roni:

Intento compilar y subir el "getMacOta.ino"

Estoy usando para el proyecto la placa "ESP32DevKitV1" en vez de la "NodeMCU ESP-32" recomendada.
Entiendo que no está probada pero te consulto porque capáz es un error conocido y facil de salvar.

El error al compilarces el siguente: 

**exit status 1
'class WiFiClass' has no member named 'STA'**

Y marca resaltada la línea 21 del scketch que dice:   **WiFi.STA.end();**

Si hay algo que pueda probar te agradezco.
Gracias

ronibandini, 2025-02-10:
Ese error suele aparecer cuando está mal seteada la placa. Fijate por las dudas que no seleccionaste un ESP8266 en la placa del Arduino IDE.

fercava, 2025-02-10:
Tengo seleccionada la **DOIT ESP32 DEVKIT V1**
Gracias

ronibandini, 2025-02-10:
En algunos foros recomiendan borrar 

C:\Users\XXX\Downloads\arduino-XXX-windows\arduino-XXX\libraries\WiFi
y
C:\Users\XXXXX\Documents\Arduino\libraries\libraries\WiFi

Y reiniciar el IDE

fercava, 2025-02-10:
Traté de actualizar diversas cosas y ahora tengo un error nuevo: 
Estoy usando IDE 1.8.16 con win10 de 32 bits...

**Esta versión de C:\Users... esptool.exe no es compatible con la versión de Windows que está ejecutando. Compruebe la información de sistema del equipo y después póngase en contacto con el anunciante de software.
exit status 1
Error compilando para la tarjeta DOIT ESP32 DEVKIT V1.**

ronibandini, 2025-02-11:
https://github.com/espressif/arduino-esp32/issues/6012

closed 2025-02-18 by ronibandini