ronibandini/tallerPocketGone #16: no funciona

Astrid97, 2025-02-11:
Buenas, necesito ayuda. En el monitor serie dice que esta en un bucle de reinicio. 

[image]

La led no prende, es por el bucle o tambien hay otro error con eso? 
probe enchufarlo en una notebook para que solo funcione como fuente y tampoco prende el led ni apaga equipos bluetooth.
Probare comprando otro cable usb. Pero ayer le subi otros codigos (como prender intermitentemente la luz azul de la placa) y funcionaba. Incluso pude hacer todos los pasos. Hoy cargue el .bin y no me fije que decia el monitor serie, se pudo haber subido con algún error que hace que funcione mal?

ronibandini, 2025-02-11:
Según parece se está reseteando el ESP32 y no llega a ejecutar el software. Podés probar desconectarle el circuito de led y botón y verificar si ahí avanza. Si tampoco avanza probá volver a cargar el getMacOta.ino y luego volve a intentar la carga del .bin

Astrid97, 2025-02-11:
lo probe desconectandole la antena, con el boton y la led conectados, deja de reiniciarse. Prende el led y realiza la accion pero no apaga un bluetooth. Deberia funcionar sin la antena?

[image]
[image]

[image]

Astrid97, 2025-02-11:
volvi a cargar el getMacOta.ino y el .bin, conecte todo de nuevo, sigue reiniciándose. Nose que puede ser

ronibandini, 2025-02-11:
En la pantalla anterior el software se ejecuta normalmente. Si al conectar el módulo de RF se cuelga yo revisaría ese cableado. Algunas personas confunden la orientación y conectan todo en espejo.

Astrid97, 2025-02-11:
Los cables estan bien conectados, pero tal vez están fallados (los compre en un lugar dudoso), el problema definitivamente es cuando conecto la antena. Si conecto el ESP32 solo los mensajes son correctos 
Mañana comprare nuevos cables, y si no es eso, compro otra placa con antena. Cualquier cosa escribo de nuevo. Gracias 

[image]

closed 2025-02-16 by ronibandini