ronibandini/tallerPocketGone #19: El arduino no ve los puertos USB

Elsebito007, 2025-02-16:
Durante la instalación del Arduino, me apareció esta imagen

[image]

No lo instalé. Estoy en una PC prestada y en ningún lugar de las instrucciones se mencionaba esa pantalla.

Seguí el resto de las instrucciones y cuando quise subir la librería a la placa me apareció esta imagen

[image]

En donde el Arduino no me reconoce ningún puerto, pero la placa está conectada al USB.

¿Cómo sigo?

¡Gracias!

Elsebito007, 2025-02-16:
La computadora parece reconocer que hay algo conectado en el puerto USB, pero el Arduino no lo ve:

[image]

ronibandini, 2025-02-16:
Dpinst_amd64.exe instala los drivers para las placas Arduino 
En este punto probá si instalando el driver del CP2102 permite reconocerlo https://www.silabs.com/developer-tools/usb-to-uart-bridge-vcp-drivers?tab=downloads

closed 2025-02-18 by ronibandini