ronibandini/tallerPocketGone #20: No silencia la música, solo la entrecorta

elrodry16, 2025-02-22:
Hola Roni, me pasa que cuando activo el dispositivo no corta la música por completo, solo hace interferencias. Ya probe en diferentes 
posiciones y distancias. Que podría ser?

elrodry16, 2025-02-22:
https://github.com/user-attachments/assets/26725a9f-fa0e-45c0-80f6-83f089a7e3ef

ronibandini, 2025-02-22:
No se ve en el video adónde está ubicado el teléfono emisor. Si el teléfono está muy cerca del parlante puede ser complicado cortar del todo el audio. Podés probar con otros parlantes y ubicaciones de los tres elementos. También se puede reemplazar la antena por otra de router sma macho más grande.

elrodry16, 2025-02-23:
> No se ve en el video adónde está ubicado el teléfono emisor. Si el teléfono está muy cerca del parlante puede ser complicado cortar del todo el audio. Podés probar con otros parlantes y ubicaciones de los tres elementos. También se puede reemplazar la antena por otra de router sma macho más grande.

El teléfono es el que está filmando. Probe con otros telefonos y funciona de la misma manera, también probé con unos auriculares BT que tengo y los mismo. Me queda probar lo de cambiar la antena

ronibandini, 2025-02-23:
Quizás vino fallado el amplificador del módulo de Radiofrecuencia. Comprobá también que la antena se encuentre bien enroscada y que tenga conector macho.

closed 2025-03-05 by ronibandini