ronibandini/tallerPocketGone #23: No estaría funcionando

27TomasLeonel, 2025-03-07:
Buenas, ya realice toda la parte del software pero al momento de conectar los cables no ha funcionado
Ya revise el monitor serial y no me arroja ningún tipo de mensaje
No sé qué podría ser

[image]
[image]

El circuito del botón lo puentee utilizando uno de los cables que me sobro y la placa que tengo es distinta de la que usan en el modelo aunque las numeraciones de las conexiones coinciden solo que pone D en vez de GPIO y 3V3 en vez de 3.3V, esto lo nombro porque entiendo que es lo único distinto que tengo 
Leí en el foro que para tratar de alimentarlo directo debía conectar el 3.3V de la placa de la antena en la conexión de VIN pero al probar eso tampoco resulto
Así que no sé que más probar para que funcione 
Desde ya muchas gracias

ronibandini, 2025-03-07:
Hola Tomás. Asumo que completaste la primera parte de cargar el getMacOta.ino y luego pudiste subir el software .bin

Probá en principio sacar el circuito de botón y led. Conectá la placa con el cable USB a la computadora y abrí el monitor serial del Arduino IDE en 115200. Deberías ver mensajes de diagnóstico. Si no es así es porque el software no está cargado.

27TomasLeonel, 2025-03-10:
Buenas tardes, en efecto el software no se había cargado, no sé porque pero de todos modos ya pude cargarlo y ahora si prende correctamente, solo resta probarlo
Desde ya muchas gracias

ronibandini, 2025-03-10:
Excelente Tomás. Cuando lo termines manda alguna foto o video. Abz

closed 2025-03-10 by ronibandini

27TomasLeonel, 2025-03-28:
Buen día, ya pude probarlo y funciona de 

https://github.com/user-attachments/assets/854ae70c-5f74-401e-bc26-2edd83606966

Muchas gracias por el curso y el asesoramiento

27TomasLeonel, 2025-03-28:
Ahora me gustaría incorporarle un amplificador para cuando se alejan de mi ventana pero sigo escuchando su ruido como si los tuviera al lado. Cuál me recomendarías? O qué me recomendarías considerar antes de comprar uno? Puedo amplificar la señal con un modem que tengo sin usar?
Desde ya muchas gracias

ronibandini, 2025-03-28:
Podés mejorar el alcance con una antena superior y en teoría es posible colocar un amplificador de 2.4 entre el Pocket y la antena. Me acaba de llegar un amplificador. Si veo que funciona luego lo recomiendo.

27TomasLeonel, 2025-03-30:
Muchas gracias, estaré atento a tu recomendación. De igual modo te mostrare si lo amplifico.
Que estés bien.