ronibandini/tallerPocketGone #24: No aparece en el listado de las redes

sweetrevenge33, 2025-03-09:
Hola Roni, al inyecto el .ino... llega al 100% parece que se hace reseting pero... no puedo obtener el MAC ya que parece que carga que no aparece en el listado de las wifi:

Me da este mensaje el serial monitor:

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4688
load:0x40078000,len:15516
load:0x40080400,len:4
load:0x40080404,len:3196
entry 0x400805a4

ronibandini, 2025-03-09:
Si tenés conectado el circuito de led y botones, desconectalo y probá nuevamente qué sale en el Monitor Serial

sweetrevenge33, 2025-03-12:
solo está conectada la placa al pc

ronibandini, 2025-03-12:
Si el software getMacOta.ino subió al 100% y te muestra este mensaje en el serial monitor es probable que la placa no esté funcionando bien

ronibandini, 2025-03-12:
Probá primero repetir el procedimiento: subir getMacOtaIno, subir luego el software .bin y sin led ni botones verificar el monitor serial. Si así no funciona, algo pasa con la placa

closed 2025-03-21 by ronibandini