ronibandini/tallerPocketGone #26: No Corta la Musica

Italiano14, 2025-03-13:
Hola que tal!!, Le comento, el dispositivo tiene todo perfectamente echo, los cables, la conexión, el .bin, el código cargado; Inclusive prende la lucecita, pero el problema que presento es que no corta la música del parlante, osea se prende la luz y la música sigue sonando. Muchas Gracias!!!

ronibandini, 2025-03-13:
Hola. ¿Probaste con varios parlantes bluetooth o con uno solo? ¿A qué distancia pusiste el parlante?

Italiano14, 2025-03-14:
> Hola. ¿Probaste con varios parlantes bluetooth o con uno solo? ¿A qué distancia pusiste el parlante?

Hola!!, probé con parlante chiquito y uno grande, a una distancia media-corta, no tan lejos.
Quizas lo que puede ser que este mal conectado los cables de la antena, pero no lo veo como opción.  Muchas Gracias, disculpe las molestias!!!

Italiano14, 2025-03-16:
> Proba la fuente de alimentación. Me acaba de pasar lo mismo, con un cargador, cambie de cargador para alimentar a la plaqueta y funcionó bien.

con que fuente de alimentancion probaste??, de cuantos Volts son los cargadores?

ronibandini, 2025-03-16:
5V, 1A va bien.

closed 2025-03-21 by ronibandini