ronibandini/tallerPocketGone #27: No aparece DOIT ESP32 DEVKIT V1

execriss, 2025-03-13:
Hola roni, espero que estes bien. 

Estoy teniendo un problema con la instalación. Hice todos los pasos tal como indicas, pero al seleccionar la BOARD no me aparece la que nombras en el video: **_DOIT ESP32 DEVKIT V1_**

La placa si me la detecta en el puerto COM6 pero no encuentro la board. 

Agradecería mucho tu ayuda. Gracias

[image]

[image]

execriss, 2025-03-13:
Cabe comentar que estoy en windows 11

ronibandini, 2025-03-13:
Hola. En los issues cerrados hay varios post para cuando no aparece el Dev Kit. Fijate en https://github.com/ronibandini/tallerPocketGone/issues/9

execriss, 2025-03-15:
Ya me funcionó, tuve que cambiar de PC.

Te hago otra consulta, ya me funciona, si bien NO CORTA del todo la música, por lo menos la entre corta, hice pruebas con varios parlantes y a distintas distancias. 

**Me gustaría poder tener mucho más alcance.** 

Estaba considerando comprar esta antena para ver si puedo extender el rango:
https://www.mercadolibre.com.ar/antena-gsm-cbase-magnetico-5db-sma-macho-cable-3mts/p/MLA46749268#polycard_client=search-nordic&searchVariation=MLA46749268&wid=MLA2026130512&position=2&search_layout=stack&type=product&tracking_id=a1b2d4f3-b778-4108-aeb7-b7db09a60d8b&sid=search

[image]

O me recomendas alguna otra?

Y por último, aun no termina [El instructivo sobre extensión de alcance ](https://bandini.medium.com/pocket-gone-para-bloquear-parlantes-bluetooth-cb3d41036733#:~:text=El%20instructivo%20sobre%20extensi%C3%B3n%20de%20alcance%20(pronto%20va%20a%20estar%20listo))?

ronibandini, 2025-03-16:
Puede funcionar esa antena y ya el cableado te va a extender el alcance. También una buena antena de router funciona. Luego, otra opción que podés tener es meter un amplificador de señal entre el Pocket y la antena. Tenía que llegarme uno por correo para escribir el tutorial, pero está demorado.

execriss, 2025-03-16:
Perfecto, voy a probar con esas antenas para aumentar el rango. 
Muchas gracias Roni. 

Y te agradecería mucho si puedes compartir el tutorial sobre extensión de alcance una vez que lo tengas finalizado.

closed 2025-03-16 by execriss