ronibandini/tallerPocketGone #28: Amplificar señal Gone

itechnelson, 2025-03-20:
Hola! Compré el curso y ya estoy armando el Pocket, hice todas las conecciones y está funcionando bien (solo me falta hacer la cajita) . En pruebas que hice, me doy cuenta que necesito mejorar el alcance, como lo podría hacer? Se puede? 
Gracias!

ronibandini, 2025-03-20:
Hay alumnos que incrementaron bastante el alcance con una buena antena de router. Fijate que tenga conector SMA macho. Luego es posible usar antenas con cable que van a incrementar alcance también por el cableado (hasta cierto punto) La mejor opción quizás sea Pocket Gone + amplificador 2.4 + antena cableada

itechnelson, 2025-03-20:
1. Gracias por la respuesta. Ya hice lo de la antena del router, tengo una de 5 dbi, hoy hago pruebas. Con respecto al amplificador 2.4, me podrías indicar algún modelo, estoy buscando en ML pero no encuentro nada.

ronibandini, 2025-03-21:
No podría recomendar uno en particular, pero fijate que sea de 2.4 y que tenga conectores SMA para meter al medio de Pocket y antena. Cuando logre probar algunos publico la info.

closed 2025-03-21 by ronibandini