ronibandini/tallerPocketGone #30: Funcionamiento luego de cargar el BIN

pablo7913, 2025-04-01:
Buenas tardes, 

Realice el proceso de configuracion de la placa, desde el punto de vista de los pasos a seguir, correctamente. 

Logre ver la red wifi con la MAC Address, y luego de recibir el archivo BIN, lo pude cargar correctamente a la placa. 

Hoy al realizar la conexion del switch, led y placa,  no me prende el led al pulsar el boton. Revisando, tampoco me aparece la red wifi de la placa. 

Queria saber si es correcto que la red wifi deje de aparecer luego de cargar el bin, y en caso de haber fallado algo en todo el proceso, puedo volver a cargar todo a la placa y arrancar de 0?

ronibandini, 2025-04-01:
Hola. Una vez cargado el bin ya no debería aparecer el wifi de la placa. Si no prende el LED al pulsar el botón es probable que haya algún error en el circuito del botón-led o bien se encuentre mal conectado a los pines. Revisá esa parte y si no encontrás nada podés abrir el monitor serial y ver los mensajes de diagnóstico que aparecen.

DannBF, 2025-04-24:
Hola!

Me está pasando algo parecido, pero imagino que el problema será otro.

He terminado de montar la parte electrónica, pero al conectar el Pocket Gone a una fuente de alimentación, el LED no se enciende ni detiene ningún speaker.
He repasado la instalación de los cables varias veces, siguiente el video explicativo y creo que está todo correcto.
También he comprobado que el LED esté conectado correctamente, comprobandolo varias veces en el video y contrastando con los problemas que ha compartido gente aquí.

Después de haber cargado el bin y conectar toda la electronica, si conecto la placa al ordenador, no encuentro la red WIFI como si pasaba antes de cargar el .bin. Quiere decir esto que el software está instalado correctamente? Podría repertir los pasos de instalación del software para asegurarme de que no hay problemas en esa parte?

Adjusto fotos de la instalación de la placa por si alguien puede detectar algún fallo.

Gracias de antemano! 

[image]
[image]
[image]
[image]

ronibandini, 2025-04-24:
Un vez cargado el bin ya no aparece el WiFi. Eso es correcto. Lo que podés hacer es conectar la placa con el cable a la computadora y abrir el monitor serial del Arduino IDE: Ctrl + Shift + M (Windows/Linux) or ⌘ + Shift + M (macOS)

Ahí te van a aparecer mensajes de diagnóstico. Una posibilidad es que esté mal el MAC address y que por eso no se inicie el dispositivo.

DannBF, 2025-04-24:
Hola @ronibandini. Gracias por tu rápida respuesta.

Me sorprendería mucho que te enviase mal el MAC, ya que has hecho mucho hincapié en que tiene que ser el correcto y en caso de que no sea así el nuevo .bin tendría un costo adicional. Lo comprobé varias veces antes de rellenar el form de google.

Te adjunto pantallazos después de haber abierto el Arduino IDE y usar Comando + Shift + M. Parece que hay algo mal. Probe varias veces y este es el resultado:

[image]
[image]

Repasando el segundo video del montaje del Pocket Gone, me he dado cuenta que compartes una imagen que hace mención a dos LEDs. En mi placa, solo se enciende uno. Deben estar los dos encendidos?

[image]

[image]

Gracias por tu ayuda

ronibandini, 2025-04-24:
Probá cambiar los baudios del serial monitor a la derecha, de 9600 a 115200

DannBF, 2025-04-25:
Hola Roni. Cambié los baudios a 115200 y le di a upload, cierto?
El Pocket Gone sigue con el mismo comportamiento.

En Arduino Ide, usando el Command + Shift + M, no me muestra ningún mensaje de error. Adjunto pantallazo después de usar el shortcut.

Qué puedo hacer a continuación?

[image]

ronibandini, 2025-04-25:
Según parece achicaste la ventana del serial monitor y por eso no se ven los mensajes de diagnóstico. Llevá el mouse a la línea divisora y levantala para que el código fuente se achique y los mensajes del serial monitor aparezcan.

DannBF, 2025-04-25:
Ok, gracias. Adjunto captura.

 
[image]

DannBF, 2025-04-25:
volviendo a usar el short cut, esta vez encuentro un diagnostico que parece infinito. El codigo parece repetirse.

[image]

ronibandini, 2025-04-25:
La placa se está reiniciando. Probá desconectar los cables de la antena y la botonera y repetir el serial monitor. Si ahi la placa se inicia normalmente puede ser un problema de cableado o de la placa nrf.

ale92h, 2025-05-12:
Buenos días. En Mercadolibre se consigue la placa Nodemcu ESP32 en lugar de la ESP32s. Hay alguna diferencia? Esta también sirve? Muchas gracias

[image]

ronibandini, 2025-05-12:
Sirve también. Por las dudas confirmá que los pines se encuentren en la misma ubicación antes de conectar los cables.

DannBF, 2025-05-12:
Hola @ronibandini , disculpa la demora en contestar, he estado fuera de casa por trabajo.

Acabo de retomar el proyecto y he seguido tus pasos: desconectar antena, botonera, y repetir el serial monitor. Esto es lo que me aparece en la pantalla (y que no se interpretar). Qué puedo hacer como siguientes pasos? Gracias.

[image]
[image]

ronibandini, 2025-05-12:
Por las dudas probá repetir el primer paso: cargá el getMacOta.ino y subile nuevamente el .bin Si el bin está bien cargado y la placa se reinicia sin tener la antena ni el circuito de botones conectado, es probable que haya venido mal el ESP32.

closed 2025-05-18 by ronibandini