ronibandini/tallerPocketGone #37: Problemas en el funcionamiento luego del montado

gustavoheinze, 2025-05-27:
Buenas tardes. Seguí el paso a paso para cargar el software y el montado de los cables.

Si bien en el esp32 los pines tenian otros nombres, busque una guia que me indicaba cual era el que correspondia. (de hecho en otro post vi alguien que tenia uno igual y funciono)

[image]

Aparentemente esta todo ok, pero cuando lo voy a probar el dispositivo no funciona.

Esto es lo que me sale en el log del IDE cuando lo activo.

[image]

lo que veo diferente es que al esp32 solo se le prende una luz.

[image]

y el nrf24l01 no tenia indicado que era cada pin. Pero seguir el modelo que se ve en el video y creo haber conectado todo como corresponde.
También tiene como esa pasta blanca que no se si es normal, ya que no estoy familiarizado con los componentes electrónicos.
Revise la placa y en un chip dice el modelo: nrf m 24L01+ 1839 ah

[image]

Gracias por la ayuda que puedan darme

ronibandini, 2025-05-27:
Hay servicios de impresión 3d muy baratos. Ambas piezas deberían costar u$3 a u$5 

Si abrís los archivos gcode con algún programa tipo [Slicer](https://slic3r.org/) creo que te aparecen las medidas, pero si vas a fabricar tu propio gabinete aprovechá para meter creatividad y hacer algo diferente.

closed 2025-05-27 by gustavoheinze

vacardozo, 2025-07-19:
Buen día @ronibandini habrá forma de saber si el módulo NRF24L01 funciona?

No tuve oportunidad de comprar el que recomiendas y compré este
https://es.aliexpress.com/item/1005008965761071.html

La consola muestra logs y el timer funciona sin problema, los cables los he comprobado muchas veces, he probado desde varios celulares a 2 parlantes, uno SRS-XB33 SONY y el otro es un adaptardor bluetooth conectado a un equipo con puerto usb, tampoco funciona conectando celular al carro 😔

Gracias por cualquier ayuda.

ronibandini, 2025-07-19:
Si el monitor serial no muestra un problema de comunicación con la placa de rf, entonces está bien cableado y quizás sea de mala calidad o un malfuncionamiento del módulo. Antes que nada igual probá con algún otro parlante bluetooth cerca.