ronibandini/tallerPocketGone #41: Consulta: Autoencendido

ale92h, 2025-07-04:
Buenos días, el Pocket funciona muy bien y me gustaría ver una posibilidad de sistematizar su uso.

En mi caso el sonido viene siempre del mismo lugar (es el mismo vecino que pone música fuerte a la siesta los domingos) y quisiera hacer una especie de "Torre Gone", dejando fijo el dispositivo y que se pueda encender a distancia. 

Para esto necesitaría que el dispositivo se active solo ni bien recibe alimentación, sin depender de un botón que lo active. Esto se puede? Ya que la parte de alimentación en 220 es donde puedo maniobrar los elementos para encenderlo a distancia (en continua nunca se ponen cables largos por la caída de potencial). 

También estoy experimentando con una antena mejor para más alcance y si logro armar la Torre Gone con alcance mejorado, les paso la receta y video. 
Desde ya, muchas gracias!

ronibandini, 2025-07-04:
Lindo proyecto. Cuando lo termines compartinos foto y video. Para el autoencendido probá desconectando el botón y conectando un jumper entre el pin 25 y GND.

ale92h, 2025-07-04:
Perfecto, muchas gracias!