ronibandini/tallerPocketGone #44: No me reconoce la placa

lealucard, 2025-07-13:
Probe con 2 pc, el cable es de datos, en distintos puertos y no me aparece dejo imagenes para ver si alguien me puede dar una mano por favor!, les dejo una captura de como me figura en dispositivos "CP2102 USB to UART..."

[image]
[image]
[image]
[image]
[image]

lealucard, 2025-07-13:
leyendo los distintos post lo pude solucionar, tuve que instalar manualmente todo pero a la hora de compilar me sale el siguiente error 

[image]

ronibandini, 2025-07-14:
En ocasiones ese error se soluciona poniendo manualmente al ESP32 en Flash Mode. 

Cuando el IDE dice Connecting, presionar y mantener el botón BOOT del ESP32 y luego presionar el botón EN un segungo. Soltar EN y soltar BOOT.

ronibandini, 2025-07-14:
Buenísimo. Cuando lo termines con algún gabinete compartinos unas fotos.

closed 2025-07-20 by lealucard