ronibandini/tallerPocketGone #45: Consulta Android TV

lealucard, 2025-07-20:
Como dice el titulo del hilo, tal vez sea tonta mi consulta, pero si transmito youtube por bluetooth desde mi celular a mi tv android el gone lo deberia detener tambien? (o bloquear), desde ya mucha gracias por despejarme de la duda... porque si es asi algo hice mal en la conexion o no me estaria funcionando.

lealucard, 2025-07-20:
Proyecto Final (muchas gracias roni) sin conocimientos de arduino y de electronica (solo un poquito de 3d) se logro esto:
[image]
[image]
https://github.com/user-attachments/assets/96b73aff-7781-4831-afeb-c0efae088417
P.d: dejo captura del circuito por ahi a alguien le resulta de ayuda mientras se ve el video.

ronibandini, 2025-07-20:
Muy bueno. Gracias por compartirlo.

closed 2025-07-20 by ronibandini