ronibandini/tallerPocketGone #48: Pocket gone 3

lealucard, 2025-08-06:
Hola roni! podrias aclaras las diferencias entre el pocket gone 1 y el nuevo que sacaste hace 3 dias, y si existe la posibilidad de que quienes compramos el 1, tengamos algun descuento en conseguir el 3, muchas gracias!
[image]

lealucard, 2025-08-06:
muchas gracias espero ansiosoo!!!! si haces un combo con el de la antena un golazo, porque voy a probar creo que a mi tambien me sucedio que la antena que me mandaron no es la correcta.

hugo-montero2, 2025-08-07:
Yo compre el 1 el dia que lanzaron el 3 :( so sad! Pregunta... Son los mismos componentes? solo cambia las mejoras en el software?

lealucard, 2025-08-08:
esperamos ansioso la promo roni!