ronibandini/tallerPocketGone #49: Software para más de 1 tarjeta

Caverocr, 2025-08-06:
Hola Rony,

Acabo de comprar el taller, en los links de compra, vienen varios componentes por paquete.  Ya vi el video del taller y hay que llenar un formulario con la MAC address de la placa, se puede solicitar software para más de una placa? 

Gracias y saludos,

Caverocr, 2025-08-07:
@hugo-montero2 , yo compré el Pocket Gone 3, minutos antes de publicar la consulta.

hugo-montero2, 2025-08-07:
Ufff, yo me acabo de enterar que salio el 3, entre para continuar con los pasos, por que apenas me llego la placa. Seran los mismos componentes? No se que tan diferente sean.

ronibandini, 2025-08-11:
La versión 3 usa otra placa ESP y otro módulo de radiofrecuencia. Sin embargo sería posible reaprovechar la placa ESP del 1 sobre el gabinete viejo. El taller incluye el software para una placa.