ronibandini/tallerPocketGone #50: Archivos para imprimir el gabinete en 3D

tincho40, 2025-08-07:
Hola Roni, soy Juan Martin, acabo de comprar el taller virtual para armar el Pocket Gone. dos consultas para hacerte:

1ra, podras pasarme los archivos para imprimir en 3D el gabinete y sus partes ?
2da, recomendas algun proveedor de electronica local en donde pueda conseguir los componentes??

desde ya muchas gracias !!!

ronibandini, 2025-08-07:
Hola Juan. Los archivos 3d se entregan junto al software bin una vez reportado el MAC address de la placa. Para recomendar proveedor de electrónica ¿en qué país estás?

ronibandini, 2025-08-07:
El ESP lo vas a encontrar en Candyho. El módulo de radiofrecuencia hasta donde sé no se consigue en negocios de Argentina

tincho40, 2025-08-10:
Roni, ahi complete el formulario con la MAC del ESP. otra consulta, si le armo otro poket para un amigo, como es el tema para mandarte una segunda MAC ?

ronibandini, 2025-08-10:
Si querés armar un segundo dispositivo solo tenés que encargar el software y después podés reportar el segundo MAC desde el form https://www.patreon.com/RoniBandini/shop/software-bin-pocket-gone-1094523

closed 2025-08-10 by ronibandini