ronibandini/tallerPocketGone #58: No funciona, no prende led, monitor serie con simbolos.

chotonito, 2025-08-17:
Hola!!
No activa el led, ni tampoco la antena, paso a detallar:
La polaridad del led es la correcta y el pulsador funciona al igual que el led.
Revise y reconecte todo tres veces, tanto la antena como el pulsador, para descartar error de cableado (cables cuya continuidad probe de placa a placa, asi tambien berificaba contacto de fichas).
Reinicie la SP32 a fábrica e hice todos los pasos nuevamente hasta cargar el bin correctamente desde http.
Cuando le conecto la alimentacion, automaticamente interfiere con la señal wifi de mis camaras, esto lo hace sin tocar el pulsador. Pero no interfiere la señal del BT del parlante. 
De todos modos, insisto, no prende el led al tocar el pulsador y el log tira simbolos (indicando que algo esta mal en la programacion)
Obivamente tambien probe puente con cable desde GND a P33, a modo de reemplazo del pulsador

Adjunto imagenes.

[image]
[image]
[image]
[image]
[image]

ronibandini, 2025-08-17:
[image]

Según parece hay algún error en el circuito botón-led. Confirmá los pines y la dirección del LED. Para visualizar el Serial Monitor tenés que cambiar los baudios a 115200 en el drop down de la derecha. Una vez que puedas leer los mensajes del serial monitor, al presionar el botón deberías ver la activación y el timer.

chotonito, 2025-08-17:
gracias, por la respuesta!!!!
En la foto no se aprecia, pero esta conectado tal cual el grafico que compartis, pasa que hay un cable gris claro y uno blanco que en la foto no se diferencian. De todos modos, con un cable hice contacto entre el 33 y gnd para obiar el pulsador (el cual igual esta probado)
Lo de los 115200, no lo sabia, pero sigue sin andar, adjunto log

[image]

ronibandini, 2025-08-20:
Quizás dejaste conectado el cable del pin 23. Remové ese cable y debería funcionar con el botón por lapsos de 50 segundos.

chotonito, 2025-08-20:
esta conectado el pulsador, no hay nningun puente
Pasa tal cual describí, lo detallo un poco mas: 
si lo pulso una vez pasa de Jumper enabled: 20 a 23 (un ejemplo) e interrumpe la señal un segundo
si lo mantengo precionado corre de Jumper enabled: 10 a 50, ahi interrumpe la señal mienras este precionado, lugo el parlante suena de nuevo.
 Despues reviso todo nuevamente y subo un video.

chotonito, 2025-08-21:
Habia visto un par de videos que precionaban una vez y listo. 
Este es de su canal si no me equivoco.
https://www.youtube.com/watch?v=-nWwI6LqKME
Que lastima, no me estaria sirveindo. Gracias!

ronibandini, 2025-08-22:
Me alegro. Si filmás un video silenciando un parlante en condiciones reales compartilo y te paso el tutorial de la antena externa