ronibandini/tallerPocketGone #63: Duda compatibilidad con distintos componentes

metratiko2023-max, 2025-08-20:
Tengo la duda de si puedo reutilizar una placa esp32 de 38 pins que tengo ya por casa con el chipset CP210X o es obligatorio comprar la que tu recomiendas en el taller y la otra duda es si puedo utilizar un nRF24L01 SPI PA LNA que tambien tengo por casa ya que no dispongo ahora mismo de el E01-ML01DP5 aunque soy consciente de que puede variar algo su potencia final. gracias

ronibandini, 2025-08-20:
Podés usar la placa de 38 pines, pero vas a tener que montar un circuito de led y botón conectados a D2 y D4 ya que el diseño original usa led y botón on board y no van a coincidir en GPIO ni ubicación.

Y podés usar el nRF24L01, pero su alcance y confiabilidad son menores al módulo recomendado.

metratiko2023-max, 2025-08-21:
Vale genial, alfinal lo e pedido todo nuevo. gracias !

ronibandini, 2025-08-21:
Cuando lo tengas terminado con el gabinete no olvides compartir fotos o video

closed 2025-08-21 by ronibandini