ronibandini/tallerPocketGone #68: La comoputadora no me detecta la placa

rafagallerano-eng, 2025-08-25:
**Version de Pocket Gone:**
 - Version 1 (tiene LED y botón frontal) o Version 3 (el botón está al dorso)

**Describir el problema**
Ejemplo: armé todo de acuerdo a las instrucciones y no prende la luz al tocar el boton

**Subir imagenes o videos**
Copiar la imagen con el video y pegar

rafagallerano-eng, 2025-08-25:
Enchufo la placa pero no computadora no me la detecta no me sale ningun dispositivo conectado

ronibandini, 2025-08-26:
¿Pocket Gone 1 o Pocket Gone 3? ¿Instalaste los drivers? ¿Buscaste otros problemas similares en https://github.com/ronibandini/tallerPocketGone/issues?q=is%3Aissue%20state%3Aclosed?

rafaelgallerano-cpu, 2025-08-26:
Es el pocket gone 3 y si instale los drivers y aún así no me reconoce . Tengo windows 11 no sé si eso será un problema . Busque el driver para el 11 y es el mismo archivo.

Cmedina19, 2025-08-28:
> Enchufo la placa pero no computadora no me la detecta no me sale ningun dispositivo conectado

me paso lo mismo, solo cambie de cable USB por uno de mejor calidad y funciono.