ronibandini/tallerPocketGone #89: Pocket Gone 3 lo probé y no pasa nada

Lara3737, 2025-09-18:
Pocket Gone 3
Hola a todos, armé el circuito, ya chequè las conexiones y enciende, primero el botón rojo y cuando pulso de nuevo enciende tambien el led azul, si vuelvo a pulsar, se apagan los dos y si vuelvo a pulsar se encienden ambos led, rojo y azul, sali en auto a probar en la calle y aunque me estacionaba enfrente del negocio no pasó nada... que podría ser?, que tuviera mala suerte y que el celular estuviera cerca?, que fuera USB o WiFi? :(

[image]
[image]

**Subir imagenes o videos**
Copiar la imagen con el video y pegar

ronibandini, 2025-09-18:
Tras el montaje, para determinar si funciona deberías probar en principio con un parlante bluetooth propio.

Lara3737, 2025-09-19:
Hola, tuve un error de cableado  y me fijé después de enviar las fotos acá, sí funcionó!!!! en casa y puse a prueba lo del acercamiento del emisor y pues ahí no funciona, tal como describes, cuando tenga oportunidad te envío el video.
Por otro lado, aparte de la lata de duraznos, que tipo de antena sugieres? hay direccional y omnidireccionaL y que modelo comercial podría ser? 
Gracias por el apoyo

ronibandini, 2025-09-19:
Excelente Lara. Podés probar con antenas Yagi o dipolo, pero probá antes con la lata de duraznos que no cuesta nada.

ronibandini, 2025-09-25:
El Peach Cantenna se opera con una sola mano y es disimulable. Sino podés probar con Yagi o Dipolo. En cualquier caso la distancia entre emisor y parlante siempre va a ser relevante.

ronibandini, 2025-09-30:
Hola Elsi, pedí el link a holapocketgone@gmail.com