ronibandini/tallerPocketGone #96: Solo deteriora la conexión. Pocket Gone 3

JairUlloa, 2025-09-26:
Pocket Gone - Version 3 

Armé todo de acuerdo a las instrucciones y confirmé cableado enre ESP32 y ML01DP5, funciona!!!!  👍  pero **solo deteriora la señal**, probé con audifonos BT, una bocina Plantronics BT y fnalmente con un altavoz JBL, 

Anexo video de la última prueba.

https://github.com/user-attachments/assets/b998f36f-2ba8-4adb-ac88-f8b6c7ffd292

Saludos!

ronibandini, 2025-09-26:
Podés probar con distintos ángulos y alturas. También verificá que la antena se encuentre bien enroscada y los cables no hagan falsos contactos.

JairUlloa, 2025-09-27:
Lo he hecho  con el mismo resultado.  Me gustaría probar ckn la antena de bote
 Podrías compartir el procedimiento? Ya mandé fotos con el gabinete.

Gracias